boarchuz/HULP #1: What are plans for documentation?

standarddeviant, 2020-06-27:
@boarchuz Firstly, thank you so much for creating this library - it looks very promising!

What are your plans for documentation? I see some nice function descriptions in the header files. I'm curious if you're planning to use a system like doxygen or spinx to autogenerate documentation.

Cheers!

boarchuz, 2020-06-28:
Hi @standarddeviant 
It's been on my mind for a while. I think it could also be a good place for general ULP programming introduction, tips and guidance.
I've just started with Doxygen + Sphinx + Breathe so expect something to appear soon.

standarddeviant, 2020-06-29:
That's great to hear! For context, I'm getting an M5Stick-C very soon and hoping to to continuously read the IMU and wake up the main cores if there's enough motion. I'm comfortable with C, but not assembly; I think HULP could greatly simplify implementing a low-power "wake-up-on-motion" in the ULP. I'll post something if I get it working.

mahesh2000, 2021-08-03:
things like `SWAPPED_TOUCH_INDEX(x)`, which exists because of a bug the ESP32 codebase; touch 8 & 9 need to be swapped.

https://github.com/espressif/arduino-esp32/issues/3584

boarchuz, 2022-09-19:
I'm not sure it's worth the trouble at this point, sorry for never completing this.
If there's anything in particular that anyone needs clarified, please don't hesitate to create a new issue.

closed 2022-09-19 by boarchuz

macvenez, 2023-01-24:
Sorry, just saw this and asking,
why not going on with development?
Are there any better alternatives?

fernsferns, 2023-01-24:
I wrote several macros inspired by this project a while ago. If I find them, I'll post them.

boarchuz, 2023-01-24:
Hi @macvenez,
I set up Doxygen briefly but got distracted by other things and never saw it though. Back then, HULP usage was even smaller than it is now so I didn't feel a pressing need for docs. It's not ideal, but almost everything is commented in the source.

I personally still use HULP almost every day! Occasionally I'll add bits here and there when I find it lacking. But it's filling a tiny tiny niche, as the ESP32 ULP has been superceded by RISCVs, so I doubt HULP will grow much in future. 

As always, if there's a bug, some feature you think would be worth adding, if something is unclear, or if you just want general advice on ULP programming or a ULP code review, please feel free to create an issue.

macvenez, 2023-01-24:
Thanks for the clarification, I'll start trying out your library to create simple projects that may require the use of the ULP.
I thought while developing I could write some documentation that would have helped me in the first place, but of course I would have shared here, maybe creating a pull request to add documentation.
If I'm not wrong the RISCV ULP can only be found in esp32-S2, right? 

Furthermore, where can I find source code for commands like `I_MOVI` `M_LABEL` `I_RSHI` which can be found in your Arduino ADC example. Am I missing something?
Thanks for everything

boarchuz, 2023-01-24:
> If I'm not wrong the RISCV ULP can only be found in esp32-S2, right?

S2, S3, C6, P4... ESP32 will be the only one with only the 'FSM' core.

> Is the development on standard esp32 considered dead?

Not entirely sure what you're asking here but ESP32 is very alive in terms of espressif/ESP-IDF support and development (it's still the best of the bunch if you ask me), and HULP only supports ESP32. Far from dead.

macvenez, 2023-01-24:
Alright thanks, I was just trying to understand correctly in order to get the right board for my next project.