boarchuz/HULP #2: PlatformIO Support

salim97, 2020-07-24:
hi
i had an issue using your library
i copyed all .h from "include" folder and .cpp from "src" folder and put them in togather in my project
when i hit compile i get errors
i'm using platformio
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

In file included from src\include\hulp.cpp:1:
src\include\hulp.h:9:10: fatal error: hal/gpio_types.h: No such file or directory
 #include "hal/gpio_types.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src\include\hulp_debug.h:4,
                 from src\include\hulp_debug.cpp:1:
src\include\hulp.h:9:10: fatal error: hal/gpio_types.h: No such file or directory
 #include "hal/gpio_types.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src\include\hulp_touch.cpp:1:
src\include\hulp_touch.h:7:10: fatal error: soc/touch_sensor_channel.h: No such file or directory
 #include "soc/touch_sensor_channel.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

boarchuz, 2020-07-24:
It's targeting IDF 4.2 which Arduino doesn't support yet. You could try any of the following:

* Wait for official ESP32 Arduino update (should be soon)
* Try the 4.x branch of ESP32 Arduino https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0 (I'm not sure if PIO makes this possible though)
* Try the 'release' branch of HULP (misnomer at this point, but it is an older version that should be compatible with 3.3 with the right config)

salim97, 2020-07-24:
i tried third solution and not all macros works so i will wait for the official esp32 arduino support
thanks for the help

closed 2020-07-24 by salim97