boarchuz/HULP #3: Example for HX711

adamcoulombe, 2020-09-25:
Hello!
Thank you for this great resource!

I have been in search of some code examples that would help me wake the ESP32 when the ULP notices a difference in weight from a HX711 sensor. 

I notice you have in this repo a threshold example for a hall effect sensor. Would I be correct in thinking that this would be a very similar approach for the HX711?

The code is a little over my head but I will try my best to contribute an example for the HX711 if I can figure it out.

Thanks again!

boarchuz, 2020-09-25:
Hi @adamcoulombe,
Sorry for the lack of documentation. 

I've added a HX711 example now. I'm not 100% sure about the calibration procedure but it seemed good enough in a few tests.

You should only need to set the two pins, and the weight of some object that you will use for calibration. Make sure there's no weight to start off with, plonk the weight down when prompted, and the rest should take care of itself.

adamcoulombe, 2020-09-28:
Holy moly! thanks for posting this.... I am going to spend some time getting this example up and running, then go from there. Thanks for adding this!

closed 2020-11-18 by boarchuz