boarchuz/HULP #6: I2C bitbanging write 16bit

robertopierobon, 2020-11-12:
Some sensors like the ADS-111x DO require 16-bit readouts.
Could you please add a macro for 16 bit write?

boarchuz, 2020-11-14:
I've been wanting something for longer reads for a while now so I whipped something up. This will allow you to point to a struct that contains all of the data (addresses, length, and an array to read to / write from):
https://github.com/boarchuz/HULP/blob/41a13b672cbb279ba8a092501c370dc747e20d12/include/hulp_i2cbb.h#L287-L454

I haven't tweaked any of the timings yet but most I2C slaves are pretty tolerant in my experience so should be fine. Let me know how you go.

rpopr, 2020-11-18:
That's fine, thank you for replying to my request.
I just adjusted at row 387 with I_GPIO_READ(sda_gpio) and added initialization of the .unused parts in the struct and everything was getting work as expected.

boarchuz, 2020-11-18:
A slightly optimised version just pushed to master.
I've decided not to fix timings, it's a lot of faffing about for something that's probably fine and not a high priority.

closed 2020-11-18 by boarchuz