boarchuz/HULP #7: ESP32-S2

sedlacekdavid, 2020-11-29:
Hi, what a great project!

Does it work with ESP32-S2 ULP ?

Thanks, keep it up, David.

boarchuz, 2020-11-29:
Hi David,
It doesn't currently work with S2 as ULP-FSM is not yet supported, and many instructions have changed. If you're using the S2 I suggest using ULP-RISCV instead anyway - it's much more capable and easier to work with (there's already a C compiler in IDF). 
HULP might add some RISCV helper functions in future (rather than FSM) but I've barely touched S2 so far.

If you're curious, I have partially implemented ULP-FSM macros in this branch - https://github.com/boarchuz/esp-idf/tree/s2-ulp-fsm - with an example here https://github.com/boarchuz/s2-ulp-fsm-demo/blob/master/main/main.c

boarchuz, 2022-09-19:
There are no plans to support any targets other than ESP32

closed 2022-09-19 by boarchuz