boarchuz/HULP #8: Add librairies in Arduino ide

laguill, 2020-12-15:
@boarchuz Firstly, thank you so much for creating this library - it looks very good !

I'm triing to build a raingauge with an esp32 to count if it is raining.
I can't find the librairie in the Arduino Ide, I tried with zip file but it is not working

Can you help me on this ?

Thank you again

boarchuz, 2020-12-19:
Hi @laguill,
I've added Arduino compatibility now, however I decided to only support ESP-IDF versions >=4.2 so it will only work if you're using the 4.2 development branch of Arduino-ESP32 for now:
https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2

boarchuz, 2020-12-19:
Until that's officially updated, if there's a particular function or something that is useful, it might be easiest to just copy what you need from HULP into your sketch

laguill, 2020-12-19:
Hi @boarchuz ,
Thank you for answering me
Sorry but I'm new to this. I just installed platformio in vs code. 
Is it working ? if not I don't know how to use the 4.2 development branch of Arduino-ESP32
is it with Arduino IDE ?

laguill, 2020-12-19:
Sorry I read too fast. 
I have successfuly installed the Hulp librairy with zip file but now I can't compile the ADC exemple

By saying only support ESP-IDF versions >=4.2 did you mean paste the Stable release link: from https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/docs/arduino-ide/boards_manager.md ?
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

laguill, 2020-12-19:
I've got this error in Arduino IDE

Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_691488 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_383486 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_691488 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_383486 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\2.6.1 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

Using board 'esp32' from platform in folder: D:\guill\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4

Using core 'esp32' from platform in folder: D:\guill\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4

Detecting libraries used...

"D:\\guill\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware [... cut ...]
                 from D:\guill\Documents\Arduino\libraries\HULP\src/hulp_arduino.h:1,

                 from D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino:9:

D:\guill\Documents\Arduino\libraries\HULP\src/hulp_compat.h:4:29: fatal error: esp_idf_version.h: No such file or directory

compilation terminated.
Utilisation de la bibliothèque HULP version 1.0.0 dans le dossier: D:\guill\Documents\Arduino\libraries\HULP 

exit status 1

Erreur de compilation pour la carte ESP32 Dev Module

boarchuz, 2020-12-20:
Looks like you're still on Arduino release 1.0.4

Try following this, switching to idf-release/v4.2 branch after cloning:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

laguill, 2020-12-20:
I tried with the 1.0.5-rc4 version but I still have an error 
Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_947014 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_254976 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_947014 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_254976 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.mkspiffs.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=D:\guill\Documents\ArduinoData\packages\esp32\tools\esptool_py\3.0.0 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

Using board 'esp32' from platform in folder: D:\guill\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5-rc4

Using core 'esp32' from platform in folder: D:\guill\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5-rc4

cmd /c if exist "D:\\guill\\Documents\\Arduino\\libraries\\HULP\\examples\\ADC\\basic\\partitions.csv" copy /y "D:\\guill\\Documents\\Arduino\\libraries\\HULP\\examples\\ADC\\basic\\partitions.csv" "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_947014\\partitions.csv"

cmd /c if not exist "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_947014\\partitions.csv" copy "D:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.5-rc4\\tools\\partitions\\default.csv" "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_947014\\partitions.csv"

        1 fichier(s) copi�(s).

Detecting libraries used...

"D:\\guill\\Documents\\ArduinoData\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-96-g2852398-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.5-rc4/tools/sdk/include/config" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.5-rc4/tools/sdk/include/app_trace" "-ID:\\guill\\Documents\\ArduinoData\\packages\\esp32\\hardware\\esp32\\1.0.5-rc4/tools/sdk/include/app_update" "-ID:\\guill\\Documents\\ArduinoData\\p [... cut ...]
D:\guill\Documents\Arduino\libraries\HULP\src/hulp_compat.h:8:6: error: #error "Unsupported IDF version"

     #error "Unsupported IDF version"

      ^

D:\guill\Documents\Arduino\libraries\HULP\src/hulp_compat.h:12:27: fatal error: esp32/clk.h: No such file or directory

compilation terminated.

Utilisation de la bibliothèque HULP version 1.0.0 dans le dossier: D:\guill\Documents\Arduino\libraries\HULP 

exit status 1

Erreur de compilation pour la carte ESP32 Dev Module

What do you meant by switching to idf-release/v4.2 branch ?

boarchuz, 2020-12-20:
1.0.5-rc4 is not compatible either, you need to be on the "idf-release/v4.2" branch:
https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2

Follow the instructions there (I also linked earlier) to install the development branch, however you must checkout to "idf-release/v4.2" after the 'Clone' step.

laguill, 2020-12-20:
Sorry to bothering you again but I still can't compile in saying  checkout to "idf-release/v4.2" after the 'Clone' step

Arduino : 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -hardware D:\guill\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_679438 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_894855 -prefs=build.warn_data_percentage=75 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware -hardware D:\guill\Documents\ArduinoData\packages -hardware D:\guill\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools D:\guill\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries -libraries D:\guill\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\guill\AppData\Local\Temp\arduino_build_679438 -warnings=none -build-cache C:\Users\guill\AppData\Local\Temp\arduino_cache_894855 -prefs=build.warn_data_percentage=75 -verbose D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino

Using board 'esp32' from platform in folder: D:\guill\Documents\Arduino\hardware\espressif\esp32

Using core 'esp32' from platform in folder: D:\guill\Documents\Arduino\hardware\espressif\esp32

cmd /c if exist "D:\\guill\\Documents\\Arduino\\libraries\\HULP\\examples\\ADC\\basic\\partitions.csv" copy /y "D:\\guill\\Documents\\Arduino\\libraries\\HULP\\examples\\ADC\\basic\\partitions.csv" "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_679438\\partitions.csv"

cmd /c if not exist "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_679438\\partitions.csv" copy "D:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\partitions\\default.csv" "C:\\Users\\guill\\AppData\\Local\\Temp\\arduino_build_679438\\partitions.csv"

        1 fichier(s) copi�(s).

Detecting libraries used...

"D:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/config" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_trace" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_update" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/asio" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bootloader_support" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bt" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/coap" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/console" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/driver" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/efuse" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-tls" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp32" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_adc_cal" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_event" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_client" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_server" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_ota" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_server" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_ringbuf" "-ID:\\guill\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sd [... cut ...]
                 from D:\guill\Documents\Arduino\libraries\HULP\examples\ADC\basic\basic.ino:9:

D:\guill\Documents\Arduino\libraries\HULP\src/hulp_compat.h:8:6: error: #error "Unsupported IDF version"

     #error "Unsupported IDF version"

      ^

D:\guill\Documents\Arduino\libraries\HULP\src/hulp_compat.h:12:27: fatal error: esp32/clk.h: No such file or directory

compilation terminated.

Utilisation de la bibliothèque HULP version 1.0.0 dans le dossier: D:\guill\Documents\Arduino\libraries\HULP 

exit status 1

Erreur de compilation pour la carte ESP32 Dev Module

laguill, 2020-12-21:
It is finally working. 
1. Start Git GUI 
2. Open Recent Repository
3. Select Branch 
4. Checkout Branch
5. Tracking Branch
6. Select origin/idf-release/v4.2
7. Wait for git to pull any changes and close Git GUI
8. Open [ARDUINO_SKETCHBOOK_DIR]/hardware/espressif/esp32/tools and double-click get.exe

boarchuz, 2020-12-22:
@laguill Thanks for posting your solution, glad to see it working

closed 2020-12-22 by boarchuz