boarchuz/HULP #10: undefined reference error in example code for simple debugging

nullptrdesaster, 2021-03-17:
Hi @boarchuz,
compiling named code returns with an error: 

main.cpp:51: undefined reference to `hulp_debug_bp_init(hulp_debug_bp_config_t const*, hulp_debug_bp_state_t**)'

as the definition in "hulp_debug.h" asks for two arguments:

esp_err_t hulp_debug_bp_init(const hulp_debug_bp_config_t* config, hulp_debug_bp_handle_t* handle = nullptr);

I am not comfortable with cpp-coding, so I kindly ask for confirmation and your help.

boarchuz, 2021-03-17:
@nullptrdesaster I just tried it here and I'm not getting any errors with the example (though the debugging stuff is a bit neglected so please let me know if you have any other issues too).
That declaration uses C++ default arguments (https://en.cppreference.com/w/cpp/language/default_arguments) so it should be fine with just the config argument provided.

Have you downloaded/cloned HULP into the 'components' subdirectory of your project?
If that's not it could you share more of the build output as well as your project directory tree?

nullptrdesaster, 2021-03-17:
Thanks for your time ..
Yes, I moved HULP into a platformio project and I'm a bit struggling with it's structure/concept. There is a folder
" ./.pio/build/debug/bootloader/bootloader/subproject/components" with subfolder  "micro-ecc/" and a single file 
"uECC_verify_antifault.o" in it. 
My platformio source directory has a  hulp/src/CMakeLists.txt which should register components. As I have the "hulp_debug.h" and friends in the "hulp/include" folder, it contains:
 "idf_component_register(SRCS "main.cpp" "hulp.cpp" 
                    INCLUDE_DIRS "../include")"
Do you think, this might be wrong?

boarchuz, 2021-03-17:
That doesn't seem quite right. Check the PlatformIO docs here for adding a component:
https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components
specifically this:

> By adding a new component to an optional folder called components in the root of your project

So you should create the `components` folder if it doesn't already exist and clone HULP such that this path exists: [project_dir]/components/HULP/src/hulp.cpp

To use the debugging example, then copy examples/Debugging/Simple/main.cpp to [project_dir]/src/main.cpp and ensure there is a basic CMakeLists.txt alongside it like this: `idf_component_register(SRCS "main.cpp")`

nullptrdesaster, 2021-03-17:
YES. 
I did as advised, now it compiles as supposed. Many thanks, now I'll dive deeper in your very interesting project.

closed 2021-03-17 by nullptrdesaster