boarchuz/HULP #14: How can I make this work with Arduino?

MateoSegura, 2021-06-24:
Hello,

I really want to use this library in my project. However, our application is already pretty far in using the Arduino framework, so if there's a way of installing this in Platform IO, that'd be appreciated.

I tried to download the source files, but I get all of these errors having to do with include paths and esp-idf versions and so on. 

I just want to make sure I'm not completely wasting my time.

Thanks for the help!

boarchuz, 2021-07-05:
Sorry @MateoSegura I must have missed this notification. 
I'm not very familiar with PlatformIO but I think the Arduino version it uses is built on IDF v3.x. Hulp requires v4.2.
I expect they'll be updating it very soon, but in the meantime it might be easiest to copy any bits you need and edit slightly for compatibility. You can drop the build error log here if you're having trouble.

uvwxy, 2021-07-23:
My progress has been as follows, building with platformio:

```
[env]
platform = espressif32
platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.2
board = pico32
framework = arduino
```

idf-release/4.2 as a requirement for hulp.

I have to set `CONFIG_HULP_FAST_CLK_CAL_CYCLES` to 1, just to see how far the compilation goes. (I'm expecting other things to go wrong until this is set to a better value). How do I figure this value out?

I then end up at:
```
*** [.pio/build/pico32_LIGHT_EDITION/firmware.elf] Implicit dependency `/Users/paul/.platformio/platforms/espressif32/builder/1' not found, needed by target `.pio/build/pico32_LIGHT_EDITION/firmware.elf'.
```

I'm not sure what to make of the `platforms/espressif32/builder/1' not found` part. Any hints?

Thanks in advance :)

boarchuz, 2021-07-24:
@uvwxy I've fixed the missing Arduino config, but the updated framework and toolchain will need to be set as per your instructions. Thanks for that!

closed 2021-09-24 by boarchuz