boarchuz/HULP #17: Arduino ide8.16 compilation error on Mac

ChassaB, 2021-10-31:
Hi, hope you can help. I am getting the following compilation error. Is it simply version incompatibility or is there something I can do to fix it?

In file included from /Users/charlesburgess/Documents/Arduino/libraries/HULP/src/hulp.h:8:0,
 from /Users/charlesburgess/Documents/Arduino/Tests/ulp_examples/hulp_pulse_count/hulp_pulse_count.ino:1:
/Users/charlesburgess/Documents/Arduino/libraries/HULP/src/hulp_compat.h:8:6: error: #error "Unsupported IDF version"
     #error "Unsupported IDF version"
      ^
/Users/charlesburgess/Documents/Arduino/libraries/HULP/src/hulp_compat.h:12:27: fatal error: esp32/clk.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

boarchuz, 2021-11-01:
Hi @ChassaB 
What version of esp32-arduino are you using? It must be >=2.0.0 (IDF version 4.x)

ChassaB, 2021-11-01:
Right,
in the Arduino IDE Boards Manager it is 1.0.6 for the ESP32 install, which is the current latest release under the board manager. My version of the IDE is 1.8.16.

Do I need to go to the ESP GIT and find ESP32 Version 2?

I'll go hunt for it now

boarchuz, 2021-11-01:
Try following the instructions for 2.0.x here: https://github.com/espressif/arduino-esp32#esp32-s2-and-esp32-c3-support
I was under the impression 2.0.0 was released but maybe it's not available in the boards manager yet for some reason.
Updating the IDE sounds like a good idea if you're still having issues.

ChassaB, 2021-11-01:
Yes, the espressif GitHub says it is but not so. I updated to the latest Arduino IDE a couple of days back so it should be up to date but I'll check again tomorrow as I'm now getting the following errors with your example ADC sketch using the ESP32 Dev Module as the selected board.

Users/charlesburgess/Library/Arduino15/packages/esp32/hardware/esp32/2.0.0/tools/sdk/esp32/include/esp_common/include/esp_err.h:122:30: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t err_rc_ = (x);                                        \
                              ^
In file included from /Users/charlesburgess/Documents/Arduino/libraries/HULP/src/hulp_arduino.h:1,
                 from /Users/charlesburgess/Documents/Arduino/libraries/HULP/examples/ADC/basic/basic.ino:9:
/Users/charlesburgess/Documents/Arduino/libraries/HULP/src/hulp.h:46:11: note: declared here
 esp_err_t hulp_configure_analog_pin(gpio_num_t pin, adc_atten_t attenuation, adc_bits_width_t width);
           ^~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
too few arguments to function 'esp_err_t hulp_configure_analog_pin(gpio_num_t, adc_atten_t, adc_bits_width_t)'

boarchuz, 2021-11-01:
Sorry, that's because I haven't updated the Arduino example.

If you look at the corresponding IDF example (main/main.cpp) you should see what needs to be changed. I'll fix it soon.

ChassaB, 2021-11-01:
Thanks. Yes I see what needs to be changed;
Replaced
    /*ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN1));
    ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN2));
    ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN3));
    ESP_ERROR_CHECK(rtc_gpio_pullup_en(PIN_ADC_PIN3));
    */
with
ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12));
ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN2, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12));
ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN3, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12));
ESP_ERROR_CHECK(rtc_gpio_pullup_en(PIN_ADC_PIN3));

I assume the additional 2 lines from the IDF example are needed as well?
ESP_ERROR_CHECK(hulp_ulp_load(program, sizeof(program), 1000UL * ULP_WAKEUP_INTERVAL_MS, 0));
ESP_ERROR_CHECK(hulp_ulp_run(0));

Now that it is working I'll have to go through and better understand the code. Haven't done anything with assembler since the 6502 in the early '80s.

Thanks for your help. If I can help with documentation or something I'd be happy to.

closed 2021-11-18 by boarchuz