boarchuz/HULP #19: Possible confrontation with some libraries

freetoair, 2022-03-02:
This is not an issue but it could indicate that there is a problem. Here's what it's all about:
  I have completed one small program that finds the min and max sine waves sampled by the ADC converter on the input GPIO34. That program is run by a ULP co-processor and everything is fine. However when I add that program to the existing ESP32 program, for WiFi connection and tracking results, I get completely irrelevant values ​​from ULP. Of course, I first suspected WiFi, and then  added the ULP program to one of the examples for the ESP32 wifi server. Except that you have to add a small 330pF capacitor between the ADC inputPIN and ground, everything else works great.
   I use several libraries to run the main program and I can't guess which one has an impact on the ULP program. To analyze this, I could somehow get data on how the ULP program and the variables arranged in memory. Is it possible to extract this from the compilation files? 

regards, freetoair

boarchuz, 2022-03-02:
Hi @freetoair,
It would be really helpful if you could strip away as much as possible until you're left with the bare minimum to reproduce the issue.
My money would be on WiFi too - does it help at all with `adc_power_acquire();` (without releasing)?
What irrelevant values are you seeing? Too high? low? 0?

boarchuz, 2022-03-02:
It might be worth going over the ULP program carefully before getting lost in the weeds too much. I'll highlight some issues to get started:

```
M_LABEL(LBL_LOOP),
I_STAGE_RST(),
I_MOVI(R3, 0),
M_LABEL(LBL_PIN2_OVERSAMPLE_LOOP),
I_ANALOG_READ(R1, PIN_ADC_PIN2),
I_ADDR(R0, R0, R1), // <<< R0 is not initialised in this loop, may have EQU_COUNT/junk
I_STAGE_INC(1),
M_BSLT(LBL_PIN2_OVERSAMPLE_LOOP, (1 << PIN2_OVERSAMPLE_SHIFT)),
I_RSHI(R0, R0, PIN2_OVERSAMPLE_SHIFT),
I_MOVI(R3, 0),
I_PUT(R1, R3, ulp_vars.valueNEW), // <<< Stores most recent value, not the average (R0) which is discarded

I_MOVI(R3, 0),
I_GET(R2, R3, ulp_vars.valueOLD),
I_SUBR(R0, R1, R2),                  // if valueNEW = valueOLD & direc = 1 goto FINISH // <<< what is direc?
```
Also FYI if the program doesn't alter R3 there is no need to reset it to 0 each time you want to GET/PUT.
Perhaps a very simple loop that reads and stores an ADC value would be a better demo?

freetoair, 2022-03-02:
Hi @boarchuz    
    This makes the whole concept a lot easier, I probably misunderstood from the comment that R3 must be reset before each GET/PUT call. Certainly some library documentation would facilitate better understanding. Now I will try to change those commands if you think that could be the cause of this behavior. .

freetoair, 2022-03-02:
These are all libraries I removed: 

//#include <SD_MMC.h>
//#include <WiFiAP.h>
//#include <ArduinoJson.h>
//#include <Wire.h>
//#include <M41T81S.h>
//#include "freertos/FreeRTOS.h"
//#include "freertos/task.h"
//#include "freertos/queue.h"
//#include "I2C_eeprom.h"
//#include <Preferences.h>

but nothing has changed. Numbers still make no sense. It does not react to voltage changes.
After that I inserted the ULP code into the ESP32 example for the SD web server and then everything is fine.

freetoair, 2022-03-03:
Hi, @boarchuz 

Solved. I'm sorry to bother you, of course it's my fault. I named two variables with only one letter difference. It cost me two days of annoyance and worst of all, I bored you too. However, that strange behavior during the connection still remains, because I can't determine it for now. Unfortunately, I still haven't been able to understand when to set R3 to zero. This coprocessor means a lot to me because the first variant was to use STM32F103 as an ADC converter, so now I have to change a lot in the program. If you are interested in how it works you can look at 45.86.59.207/index.

regards, freetoair

closed 2022-03-07 by freetoair