boarchuz/HULP #21: HULP ADC read problem while wake up

kwis2, 2022-04-27:
There is problem with ADC read values while ESP32 wakes up.
detailed description https://stackoverflow.com/questions/71987305/how-to-proper-read-adc-by-ulp-while-esp32-wake-up
In my opinion it is Arduino or HULP configuration problem while wake up. 
Are you able to exclude that HULP is not the reason?
Best regards
Kamil

boarchuz, 2022-04-28:
Hi @kwis2,
Can you elaborate on this linked comment: "Similar ESP-IDF code works without problem"?
Do you mean that the same code works using IDF+HULP? Or IDF _without_ HULP?

At this point I'm guessing the bootloader is fiddling with the ADC config briefly, perhaps for RNG. But that wouldn't explain the IDF/Arduino disparity so I'm not sure.

kwis2, 2022-04-28:
Hi @boarchuz 

I am not exactly sure what do you mean IDF + HULP. In fact it is my first project with ESP32.
"Similar ESP-IDF code works without problem" mean that I have rewriten HULP part of code into asm and compiled with ESP-IDF compiler (VSC + PlatformIO + ESP-IDF in default config). As a result I have received "pure" ADC signal while ESP wakes up. 
Below there is plot of ADC data with "saturation" I have received from continous ADC measurements with HULP and Arduino.
[image]

My HULP experiences was in VSC + PLatformIO + Arduino environement in default config. 
Best regards
Kamil

closed 2022-09-19 by boarchuz