boarchuz/HULP #32: I2C bitbang example gpio question

ssriblo, 2024-03-06:
Hello @boarchuz, thanks a lot for the great lib!

I've issue with I2C bitbang example. SCL, SDA pins are not toggled yet. I use esp32 classic SoC, pins 13, 14 as at the example. Pullup resistor connected to 3.3V power line. BTW when I did try blinking example the same pins works fine. 

Main difference is pin setup:
I2C bitbang code uses: **RTC_GPIO_MODE_INPUT_ONLY**
`ESP_ERROR_CHECK(hulp_configure_pin(SCL_PIN, RTC_GPIO_MODE_INPUT_ONLY, GPIO_FLOATING, 0));`
but blinking example uses: **RTC_GPIO_MODE_OUTPUT_ONLY**
`ESP_ERROR_CHECK(hulp_configure_pin(SCL_PIN, RTC_GPIO_MODE_OUTPUT_ONLY, GPIO_FLOATING, 0));`

I see that bitbang code uses different pin control with:
` I_GPIO_OUTPUT_EN(scl_gpio)`
` I_GPIO_OUTPUT_DIS(scl_gpio)`

but blinking code uses:
`I_GPIO_SET(scl_gpio,1)`
`I_GPIO_SET(scl_gpio,0)`

BTW, I use android/platformio framework, may it be cause of issue?

Thanks and Regards,
Sergey

boarchuz, 2024-03-06:
Hi @ssriblo,
Are you able to share your code and an oscilloscope capture of the SCL and SDA pins?

ssriblo, 2024-03-06:
[HULP_I2C_bitbang_Example.zip](https://github.com/boarchuz/HULP/files/14511563/HULP_I2C_bitbang_Example.zip)
Hi @boarchuz,  I'm very appreciate your quickly response!

 my bitbang I2C test project attached. Actually very close to your example, only platformion used

The same code but based on your blinking example works well, oscilloscope indicates pulses well.
I use oscilloscope to check `GPIO_NUM_13` pin. This pin connected with pullup resistor 2KOm to 3.3V
`#define SCL_PIN GPIO_NUM_13`

boarchuz, 2024-03-06:
Is there any serial output?
Any difference if you use different pins?

Please scope the data pin too. What do you see, exactly?

ssriblo, 2024-03-07:
> Is there any serial output? Any difference if you use different pins?
> 
> Please scope the data pin too. What do you see, exactly?
Hi @boarchuz 
I did try many different pins - 2, 13, 14, 25, 27, 33 (all from rtc-gpio list) - the same issue. Oscilloscope indicates 3V high level, with no pulses. 
Test the same pins with ordinary blink code - works, oscilloscope indicates pulses as usual
`pinMode(SCL_PIN, OUTPUT_OPEN_DRAIN);
while (1)   {
    digitalWrite(SCL_PIN, HIGH); delay(10);
    digitalWrite(SCL_PIN, LOW);  delay(10);
}`
Serial output at the program start is ok, so program and hardware work well, I'm sure. I tried two different hardware boards, my own one and dev board with the same issue.

There is one chance - to use esp-idf environment instead of platformio/arduino. But I should use platformio as default ide

boarchuz, 2024-03-08:
@ssriblo There should be frequent serial output from the loop in your setup task. If not, that means it's not even doing the WAKE instruction. Can you change portMAX_DELAY to 1000? Please post the output here.
You'll need to use `idf.py menuconfig` to configure your project, specifically the option to enable ULP.

ssriblo, 2024-03-08:
Hi @boarchuz , 
serial output below, sorry, did not write before. 
portMAX_DELAY replace to 1000, but output the same:

`19:20:16.161 > Read8: 1, Read16: 0, NACK Errors: 0, Bus Errors: 2
19:20:17.160 > Read8: 3, Read16: 0, NACK Errors: 0, Bus Errors: 3
19:20:17.166 > Read8: 3, Read16: 0, NACK Errors: 0, Bus Errors: 4
19:20:18.161 > Read8: 5, Read16: 0, NACK Errors: 0, Bus Errors: 5
19:20:18.166 > Read8: 5, Read16: 0, NACK Errors: 0, Bus Errors: 6
19:20:19.160 > Read8: 7, Read16: 0, NACK Errors: 0, Bus Errors: 7
19:20:19.166 > Read8: 7, Read16: 0, NACK Errors: 0, Bus Errors: 8`

boarchuz, 2024-03-08:
Incrementing "Bus Errors" indicates that one or both of the pins are not high level at idle each time it tries to start a transaction. You'll need to figure out why it's reading low.

Make sure you're using GPIOs 13 and 14 on your board - the labels on some boards can be confusing. Make sure nothing else is connected, only the pullups to 3V3. Double check the pin levels with your oscilloscope.

An oscilloscope capture and photo of your board setup would be useful if you want me to take a look.

ssriblo, 2024-03-11:
Hi @boarchuz ,
Thanks a lot for your patience, my fault was that sda pullup was not connected. 
I appreciate your help! 
I wish you every success!

closed 2024-03-11 by boarchuz