boarchuz/HULP #38: ADC e Hibernación

SCSELECTRONICS, 2024-12-08:
Hello, good morning. I would like to thank anyone who can help me with my code. I am currently developing a critical voltage monitor.

As a background, I explain that in one part of the code, the ESP32 continuously measures the voltage on the GPIO_NUM_34 pin, which is connected to a 10k and 1k voltage divider to measure a maximum of 20V. The purpose of this code segment is to measure the voltage, and when it drops below 6V, the ESP32 enters deep sleep mode, pre-loading the program into the ULP coprocessor.

Once in deep sleep mode, the ULP monitors the voltage using the same ADC pin. I plan to implement two measurement modes: one of almost continuous measurement of 1 minute and another of measurement every 10 minutes. In the ULP program, the idea is to measure the ADC voltage and, if it exceeds 7V (equivalent to an ADC value of 792), wake up the ESP32. If the voltage is lower, the ULP will determine which measurement mode to enter: if it is higher than 5V, it activates the 1-minute mode; if it is lower than 5V, it activates the 10-minute mode. In both cases, the ESP32 should enter hibernation mode for 1 minute or 10 minutes, respectively.

However, I am now working on a simple example to test that if the voltage exceeds the threshold, wake up the ESP32; and if it is lower, just have a delay. The problem is that the program seems to wake up the ESP32 continuously, as if it determines that the threshold has been exceeded, although in reality I am injecting a voltage much lower than the expected value into the ADC. I would be grateful to anyone who can help me with this example and the main code.

`#include "hulp_arduino.h"

#define PIN_ADC_VBAT GPIO_NUM_34
#define BATTERY_THRESHOLD_WAKEUP 792  // Ajusta según la calibración de tu ADC

void ulp_init() {
    enum {
        LBL_CHECK_ADC,
        LBL_WAKEUP_TRIGGERED,
    };

    const ulp_insn_t program[] = {
        // Leer el valor del ADC del pin de la batería
        M_LABEL(LBL_CHECK_ADC),
        I_ANALOG_READ(R1, PIN_ADC_VBAT),

        // Comparar con el umbral
        I_MOVI(R0, BATTERY_THRESHOLD_WAKEUP),  // Cargar el umbral en R0
        I_SUBI(R2, R1, R0),                    // R2 = R1 - R0
        M_BGE(LBL_WAKEUP_TRIGGERED, 1),        // Salta a despertar si ADC >= umbral

        // Si ADC < umbral, quedarse en monitoreo
        I_DELAY(4000000),                      // Pausar por 500 ms (ajusta según la frecuencia de reloj del ULP)
        I_JUMPS(0, 0, LBL_CHECK_ADC),          // Volver a leer el ADC

        // Despertar al ESP32 si ADC >= umbral
        M_LABEL(LBL_WAKEUP_TRIGGERED),
        M_WAKE_WHEN_READY(),
    };

    // Configurar el pin ADC
    esp_err_t err = hulp_configure_analog_pin(PIN_ADC_VBAT, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12);
    if (err != ESP_OK) {
        Serial.print("Error configurando el pin ADC: ");
        Serial.println(err);
        return;
    }

    // Cargar el programa ULP
    err = hulp_ulp_load(program, sizeof(program), 0, 0);
    if (err != ESP_OK) {
        Serial.print("Error cargando el programa ULP: ");
        Serial.println(err);
        return;
    }

    // Configurar un valor inicial para el programa ULP (opcional)
    ulp_set_wakeup_period(0, 500000);  // 500 ms de monitoreo

    // Ejecutar el programa ULP
    err = hulp_ulp_run(0);
    if (err != ESP_OK) {
        Serial.print("Error ejecutando el programa ULP: ");
        Serial.println(err);
        return;
    }
}

void setup() {
    Serial.begin(115200);

    if (hulp_is_deep_sleep_wakeup()) {
        Serial.println("ESP32 despertado por ULP.");
    } else {
        Serial.println("Reinicio normal.");
    }

    delay(1000);  // Breve pausa para depuración

    ulp_init();  // Inicializar el ULP

    Serial.println("Entrando en deep sleep...");
    esp_sleep_enable_ulp_wakeup();  // Configurar despertar por ULP
    esp_deep_sleep_start();         // Entrar en deep sleep
}

void loop() {
}`

boarchuz, 2024-12-08:
- I_SUBI takes an immediate value. Use I_SUBR for register arithmetic 
- M_BGE is conditional on the value of R0
- I_JUMPS' parameters are (pc_offset, comparison_value, comparison_type)
- The program must end in I_HALT else execution will continue into invalid memory after the M_WAKE_WHEN_READY

SCSELECTRONICS, 2024-12-08:
According to your corrections it would be like this, but I don't understand the I_JUMPS thing, I corrected my code and haven't tested it yet, but would it be okay this way?

`const ulp_insn_t program[] = {
    // Leer el valor del ADC del pin de la batería
    M_LABEL(LBL_CHECK_ADC),
    I_ANALOG_READ(R1, PIN_ADC_VBAT),

    // Comparar con el umbral
    I_MOVI(R0, BATTERY_THRESHOLD_WAKEUP),  // Cargar el umbral en R0
    I_SUBR(R2, R1, R0),                    // R2 = R1 - R0
    M_BGE(LBL_WAKEUP_TRIGGERED, R0),       // Salta a despertar si R1 >= R0

    // Si ADC < umbral, quedarse en monitoreo
    I_DELAY(4000000),                      // Pausar por 500 ms (ajusta según la frecuencia de reloj del ULP)
    I_JUMPS(-6, 0, 0),                     // Volver a leer el ADC (offset de -6 instrucciones)

    // Despertar al ESP32 si ADC >= umbral
    M_LABEL(LBL_WAKEUP_TRIGGERED),
    M_WAKE_WHEN_READY(),                   // Preparar para despertar al ESP32
    I_HALT(),                              // Detener la ejecución del programa
};`

boarchuz, 2024-12-09:
`I_JUMPS` is defined here for reference: https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/ulp/ulp_fsm/include/esp32/ulp.h#L980-L987
It's usually easier to use `M_BSGE` (>=), `M_BSLT` (<) or `M_BSLE` (<=) for conditional jumps based on the stage counter instead of `I_JUMPS`.
Although I see now you're not using the stage counter at all so this is unusual. Do you just want an unconditional jump to a label: `M_BX(LBL_CHECK_ADC)`?

`M_BGE` is defined here: https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/ulp/ulp_fsm/include/esp32/ulp.h#L832-L834
So in the previous `I_SUBR` instruction, you'll want the result stored in R0 so that the `M_BGE` jump can then be conditional on this result. Alternatively, you could rework the logic there to jump conditional on the ALU overflow (`M_BXF`) if the threshold is exceeded.
Personally, assuming BATTERY_THRESHOLD_WAKEUP is a constant, I would simply do this:
```
I_ANALOG_READ(R0, PIN_ADC_VBAT),
M_BGE(LBL_WAKEUP_TRIGGERED, BATTERY_THRESHOLD_WAKEUP),
```

Note also that the `I_DELAY` parameter is 16 bits so 4000000 is far too large and will not have the intended effect. HULP has some delay macros you might want to check out (eg. `M_DELAY_MS_20_60000(500)`).