boarchuz/HULP #39: ADC Basic example doesn't wake up from deep sleep

AndroidCrypto, 2025-03-21:
First - thank you for your work. It is really hard to find any examples for programming the ESP32 ULP processor.

**Error: I'm using the ADC "basic" example, but it doesn't wake up after going into deep sleep.**

This is the "stripped off" version, where the parts one and two are removed:

// Source: HULP example ADC basic
// Status: does not wake up from PIN3 (nether pin25 connected to GND or 3.3V)

```
/* ADC Example
    This example will perform some basic ADC operations with the ULP, using 3 pins:
        PIN3: Wake from deep sleep if value is below threshold
*/

#include "hulp_arduino.h"

#define PIN_ADC_PIN3    GPIO_NUM_25

// Pin 3 settings
//      The internal pullup for ADC PIN3 is enabled in this example, so the idle value is ~4095 (using default 12 bit attenuation).
//      Set the threshold just a little lower, so even a small analog change will trigger a wakeup. (Try a resistor to GND, for example.)
#define PIN3_WAKE_THRESHOLD (2000)

#define ULP_WAKEUP_INTERVAL_MS (20)

RTC_DATA_ATTR struct {
    struct {
        ulp_var_t debug;
        ulp_var_t armed;
        struct {
            ulp_var_t reason;
            ulp_var_t counter;
        } wakeup;
    } pin3;
} ulp_vars;

void ulp_init()
{
    enum {
        LBL_PIN3_WAKEUP_TRIGGERED,
        LBL_HALT,
    };

    const ulp_insn_t program[] = {
        // Set a register to 0 for use with I_GET and I_PUT
        I_MOVI(R3, 0),

        // #3 : Wake if below threshold
            I_ANALOG_READ(R1, PIN_ADC_PIN3),
            I_PUT(R1, R3, ulp_vars.pin3.debug),
            // Skip if the trigger isn't 'armed'. SoC enables this before going to sleep (see below).
            I_GET(R0, R3, ulp_vars.pin3.armed),
            M_BL(LBL_HALT, 1),
            // Subtract the threshold. If it overflows, the value must be < threshold so process the event.
            I_SUBI(R0, R1, PIN3_WAKE_THRESHOLD),
            M_BXF(LBL_PIN3_WAKEUP_TRIGGERED),
            M_BX(LBL_HALT),
            M_LABEL(LBL_PIN3_WAKEUP_TRIGGERED),
                // Store the value that caused the wakeup so SoC can inspect when it's ready.
                I_PUT(R1, R3, ulp_vars.pin3.wakeup.reason),
                // Increment the counter.
                I_GET(R0, R3, ulp_vars.pin3.wakeup.counter),
                I_ADDI(R0, R0, 1),
                I_PUT(R0, R3, ulp_vars.pin3.wakeup.counter),
                // Clear 'armed'. SoC can set it again when it goes to sleep.
                I_PUT(R3, R3, ulp_vars.pin3.armed),
                M_WAKE_WHEN_READY(),
                //I_WAKE(), // doesn't work 

        M_LABEL(LBL_HALT),
            I_HALT(),
    };

    ESP_ERROR_CHECK(hulp_configure_analog_pin(PIN_ADC_PIN3, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12));
    ESP_ERROR_CHECK(rtc_gpio_pullup_en(PIN_ADC_PIN3));

    ESP_ERROR_CHECK(hulp_ulp_load(program, sizeof(program), 1000UL * ULP_WAKEUP_INTERVAL_MS, 0));
    ESP_ERROR_CHECK(hulp_ulp_run(0));
}

void setup()
{
    Serial.begin(115200);
    delay(1000);
    Serial.println(F("ESP32 HULP ADC Basic 2"));
    if(hulp_is_deep_sleep_wakeup())
    {
        Serial.print("Woken up! PIN3: Wakeup value: ");Serial.print(ulp_vars.pin3.wakeup.reason.val);Serial.print(", counter: ");Serial.println(ulp_vars.pin3.wakeup.counter.val); 
    }
    else
    {
      Serial.println(F("not deep sleep wakeup, ulp_init"));
        ulp_init();
    }
  
    // Print some values for a few seconds
    for(int i = 0; i < 5; ++i)
    {
        Serial.print("PIN3: Value: ");Serial.println(ulp_vars.pin3.debug.val);
        delay(1000);
    }
  
    // Don't go to sleep if PIN3 is still in a triggered state
    Serial.println("Waiting for PIN 3 idle...");
    while(ulp_vars.pin3.debug.val < PIN3_WAKE_THRESHOLD)
    {
        delay(100);
    }
  
    // Tell ULP that PIN3 is now armed, and sleep
    Serial.println("Sleeping...");
    ulp_vars.pin3.armed.val = 1;
    esp_sleep_enable_ulp_wakeup();
    esp_deep_sleep_start();
}
    
void loop(){}
```

On startup, I'm connecting **GPIO pin 25 to GND** to force the sketch to wait for "releasing" the pin, with this print out on the Serial Monitor:

```
ESP32 HULP ADC Basic 2
not deep sleep wakeup, ulp_init
PIN3: Value: 0
PIN3: Value: 0
PIN3: Value: 0
PIN3: Value: 0
PIN3: Value: 0
Waiting for PIN 3 idle...
```

Then I'm disconnecting pin 25 from GND and the ESP32 goes into deep sleep as expected:
`Sleeping...`

But now the E**SP32 isn't waking up anymore from deep sleep**. The ESP32 remains in deep sleep, regardless of whether I connect pin 25 to GND (should force him to wake up) or 3.3 volts (that is '4095' meaning above the threshold and not to wake up).

I tested the sketch on Arduino 2.3.4 and used esp32 SDK version 2.0.17 (newest one in 2.x branch) and esp32 version 3.1.3 (newest one). I used an ESP32 development board with 30 pins and 38 pins for testing.

Thanks in advance for your kindly help.
Michael (AndroidCrypto)

noisetaub, 2025-03-24:
Hello AndroidCrypto, I have the same problem—I can't wake up the SoC with Pin 3, but I can see the values in the monitor. Have you been able to solve it? I have tested it with Wemos D1 Mini ESP32 and ESP32 Devkit 1.

noisetaub, 2025-03-24:
I change the lines in the ulp code, and do 2 boot with the pin 3 connected to 3,3v and gnd but it doesn't wake up

`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
PIN1: Value: 667
PIN2: Value: 352
PIN3: Value: 0
PIN1: Value: 624
PIN2: Value: 340
PIN3: Value: 0
PIN1: Value: 657
PIN2: Value: 361
PIN3: Value: 0
PIN1: Value: 636
PIN2: Value: 337
PIN3: Value: 0
PIN1: Value: 615
PIN2: Value: 321
PIN3: Value: 0
Waiting for PIN 3 idle...
Sleeping...
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
PIN1: Value: 665
PIN2: Value: 359
PIN3: Value: 4095
PIN1: Value: 622
PIN2: Value: 333
PIN3: Value: 4095
PIN1: Value: 656
PIN2: Value: 346
PIN3: Value: 4095
PIN1: Value: 610
PIN2: Value: 312
PIN3: Value: 4095
PIN1: Value: 624
PIN2: Value: 320
PIN3: Value: 4095
Waiting for PIN 3 idle...
Sleeping...
`

AndroidCrypto, 2025-03-24:
@boarchuz: Thanks for your reply, but unfortunately the new code doesn't work.

Regarding your debug logs:

**1. When it is awake, alternately connect and disconnect GPIO25 from GND while it is printing some values.**

```
12:59:13.613 -> ESP32 HULP ADC Basic 3 SO
12:59:13.613 -> not deep sleep wakeup, ulp_init
12:59:13.613 -> PIN3: Value: 0     // potentiometer connected to GND
12:59:14.587 -> PIN3: Value: 0     // potentiometer connected to GND
12:59:15.619 -> PIN3: Value: 4095  // potentiometer is disconnected from GND
12:59:16.588 -> PIN3: Value: 4095  // potentiometer is disconnected from GND
12:59:17.587 -> PIN3: Value: 0     // potentiometer i reconnected to GND
12:59:18.624 -> Waiting for PIN 3 idle...
// no sleeping...

12:59:19.203 -> Sleeping...       // after reconnection to GND
```

Behaviour: no wakeup regardless any position of the potentiometer or disconnecting from GND

**2. Replacing of code lines**

This is the full log (Core Debug Level verbose):

```
13:12:35.479 -> ets Jul 29 2019 12:21:46
13:12:35.479 -> 
13:12:35.479 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:12:35.479 -> configsip: 0, SPIWP:0xee
13:12:35.479 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:12:35.479 -> mode:DIO, clock div:1
13:12:35.548 -> load:0x3fff0030,len:4916
13:12:35.548 -> load:0x40078000,len:16492
13:12:35.548 -> load:0x40080400,len:4
13:12:35.548 -> load:0x40080404,len:3524
13:12:35.548 -> entry 0x400805b8
13:12:35.614 -> [     1][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d35a0
13:12:35.647 -> [    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d3570
13:12:35.647 -> [    26][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d3540
13:12:35.647 -> [    39][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d3510
13:12:35.679 -> [    53][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d35a0
13:12:35.680 -> [    66][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d3570
13:12:35.711 -> [    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d3540
13:12:35.711 -> [    93][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d3510
13:12:35.711 -> [   107][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x400d35a0
13:12:35.743 -> [   120][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x400d3570
13:12:35.743 -> [   133][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x400d3540
13:12:35.777 -> [   147][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x400d3510
13:12:35.809 -> [   177][D][esp32-hal-cpu.c:263] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
13:12:35.809 -> [   200][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdb60
13:12:35.841 -> [   211][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdb60
13:12:35.841 -> =========== Before Setup Start ===========
13:12:35.841 -> Chip Info:
13:12:35.841 -> ------------------------------------------
13:12:35.975 -> Flash Info:
13:12:35.975 -> ------------------------------------------
13:12:35.975 ->   Chip Size         :  4194304 B (4 MB)
13:12:36.007 ->   Block Size        :    65536 B (  64.0 KB)
13:12:36.007 ->   Sector Size       :     4096 B (   4.0 KB)
13:12:36.007 ->   Page Size         :      256 B (   0.2 KB)
13:12:36.007 ->   Bus Speed         : 80 MHz
13:12:36.007 ->   Bus Mode          : QIO
13:12:36.007 -> ------------------------------------------
13:12:36.041 -> Partitions Info:
13:12:36.041 -> ------------------------------------------
13:12:36.041 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
13:12:36.041 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
13:12:36.075 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
[... 11 lines cut ...]
13:12:36.203 -> Board Info:
13:12:36.203 -> ------------------------------------------
13:12:36.203 ->   Arduino Board     : ESP32_WROOM_DA
13:12:36.203 ->   Arduino Variant   : esp32da
13:12:36.203 ->   Arduino FQBN      : esp32:esp32:esp32da:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=verbose,LoopCore=1,EventsCore=1,EraseFlash=none
13:12:36.236 -> ============ Before Setup End ============
13:12:36.332 -> [   717][V][esp32-hal-uart.c:421] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
13:12:36.332 -> [   725][V][esp32-hal-uart.c:510] uartBegin(): UART0 not installed. Starting installation
13:12:36.365 -> [   736][V][esp32-hal-uart.c:575] uartBegin(): UART0 initialization done.
13:12:37.387 -> ESP32 HULP ADC Basic 3 SO
13:12:37.387 -> not deep sleep wakeup, ulp_init
[... cut ...]
13:12:38.350 -> PIN3: Value: 3643
13:12:39.366 -> PIN3: Value: 3645
13:12:40.370 -> PIN3: Value: 4095 // after rotating the know
13:12:41.366 -> PIN3: Value: 4095
13:12:42.383 -> Waiting for PIN 3 idle...
13:12:42.383 -> Sleeping...
```

Behaviour: no wakeup regardless any position of the potentiometer or disconnecting from GND

So I'm sorry, but there is no wakeup on any ADC measurements.

@noisetaub: kindly format the log results by using the "code" tag ("<>") in the GitHub editors header, thanks.

boarchuz, 2025-03-24:
@AndroidCrypto 
It looks like ADC is now powered down when preparing for deep sleep. The ULP is getting stuck on the ADC instruction because the peripheral is powered down in deep sleep so the measurement never completes.

**Short answer:**
You'll need to do something like this:
```
I_ADC_POWER_ON(),
I_ANALOG_READ(R1, PIN_ADC_PIN3),
I_ADC_POWER_OFF(),
```
**Long answer:**
However you'll need to be very conscious of potential pitfalls if the ULP is also going to be doing this while the SoC is awake. For example, (1) if the ADC is powered down immediately after the ULP executes I_ADC_POWER_ON then it will lock up on the ADC instruction, or (2) if the ULP powers off the ADC it may interfere with peripherals the SoC is using such as RNG or WiFi.

Ideally you would edit the ESP-IDF sleep initialisation to *not* force power down the ADC, but another option is to control the ADC via software (rather than using the ULP ADC instruction) which ensures execution can never lock up. I have this for ADC1 for your reference (note GPIO25 uses ADC2; use a free ADC1 GPIO if you want to use the below code):

```
#define M_ADC1_START() \
        I_WR_REG_BIT(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_START_SAR_S, 1), \
        I_WR_REG_BIT(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_START_SAR_S, 0)

#define I_ADC1_GET_DONE() I_RD_REG_BIT(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_DONE_SAR_S)
#define I_ADC1_GET_VAL() I_RD_REG(SENS_SAR_MEAS_START1_REG, SENS_MEAS1_DATA_SAR_S, SENS_MEAS1_DATA_SAR_S + 15)
#define I_ADC1_SET_CHANNEL(gpio_num) I_WR_REG(SENS_SAR_MEAS_START1_REG, SENS_SAR1_EN_PAD_S, SENS_SAR1_EN_PAD_S+11, hulp_adc_get_channel_num(gpio_num))

I_ADC1_SET_CHANNEL(PIN_ADC_PIN3),
M_ADC1_START(),
M_LABEL(LBL_ADC_WAIT_DONE),
I_ADC_POWER_ON(), // Continually ensure ADC is powered on until complete
I_ADC1_GET_DONE(),
M_BL(LBL_ADC_WAIT_DONE, 1),
// ADC measurement complete. Check if it's ok to power down.
I_GET_WAKE_READY(),
M_BL(LBL_ADC_KEEP_POWERED_ON, 1),
  // SoC is asleep, it's safe to power down ADC to reduce sleep power consumption
  I_ADC_POWER_OFF(),
M_LABEL(LBL_ADC_KEEP_POWERED_ON),
I_ADC1_GET_VAL(), // R0 = ADC val

```

boarchuz, 2025-03-24:
Come to think of it, another very simple option is to use 2.0.0 with an ADC1 input.

AndroidCrypto, 2025-03-24:
I'm very sorry, but none of the two options worked for me - the ESP32 isn't waking up from deep sleep. 

I changed the GPIO to 34 (ADC 1 CH 6). I don't understand your third option ("Come to think of it, another very simple option is to use 2.0.0 with an ADC1 input.") - what is "2.0.0" ??

Just a note - in two hours I will start my 2 weeks vacation on a Canary Island so I can't answer or test anything in the next two weeks. I would be very happy to get an example code with two thresholds - an upper and a lower branch. 

But: the ULP should wakeup the ESP32 when the ADC value is **outside** the thresholds, meaning: if the ADC value is below the lower and above the upper threshold (in an other issue I found an example for waking up when being inside the thresholds).

Thank you again. B.t.w., I'm planning to write an article series about ULP using HULP on medium.com, and that example could help me a lot in writing the final tutorial code.

Michael (AndroidCrypto)

noisetaub, 2025-03-24:
> Come to think of it, another very simple option is to use 2.0.0 with an ADC1 input.

I have installed version 2.0.0 of Espressif in the board manager and now it wakes up, thanks

boarchuz, 2025-03-25:
@AndroidCrypto 
When you get back, please try GPIO 32 or 33 (these have internal pullups available, unlike 34).
If that doesn't work, share your ESP32 board manager version, all code, and full serial output.
And please clarify if you need the ULP performing ADC measurements while awake too, or only while in deep sleep.
Have a good break!

AndroidCrypto, 2025-03-31:
I will try changing the ADC Pin to the ones you mentioned. 

Regarding the measurements when wakeup and/or ulp only: I need the "outer" wakeup's for voltage measurements. When the voltage drops below the lower or rises above the upper threshold the ULP should wakeup the Main processor. When the esp32 ist running I need to measure as well to give a feedback to the user.
To make it short: I need the ADC in both modes.

Greetings from sunny Gran Canaria 
Michael

AndroidCrypto, 2025-04-09:
**@boarchuz:** After a 2 week break I started with ESP32 deep sleep mode and ULP wakeup. I changed the pin to 32 or 33 with no success. Today I installed Ardino 2.3.6 with ESP32 board version 2.0.0 in another user of my Windows PC and your HULP/examples/ADC Basic sketch worked like expected (I just connected the potentiometer to GPIO34 = "pin3"), so the sketch is working on this outdated ESP32 board version. Then I used the same sketch on Arduino 2.3.6 with (current) ESP32 boards 3.2.0 - this is **not working** meaning **no ULP wakeup is triggered**.

I believe there were some major changes that causes this behaviour, but unfortunately I don't have much spare time to find out how to resolve this problem. I will check for future updates of your HULP library and leave this issue "open", but I doubt there are updates in the future because the ULP programming on ESP32 ("legacy") and ESP32-S3 is outdated due to the new RISC V processor generation (e.g. ESP32-C3, -C6 and -H2).

Thanks for your kindly help so far, kind regards
Michael